Robo3D/roboOctoprint #4: Add a non-raft slicing profile for slicing through dashboard

CobyKabili, 2017-04-10:
Got an email from education stating they did not want to print with a raft when slicing from the web dashboard when they had chromebooks only. 

@AllenMcAfee should we add in all of the profiles without a raft as well so they can pick from them? That would bring a total of 8 profiles to the C2 octoprint. High, Medium, Low, Ultra High with and w/o raft.

CobyKabili, 2017-06-23:
@Ximidar can you add profiles for with and without a raft for web dashboard slicing? This was a request from sales for schools. Remember, without a raft needs a bigger first layer width. Do you need @JasonBarretto to create these for you?

Ximidar, 2017-06-23:
@CobyKabili Yes it would be better for @JasonBarretto to create them, He has a better grasp of cura settings than I do