Robo3D/roboOctoprint #13: Access Control copy edits

peries, 2017-11-20:
@victorevector @AllenMcAfee for your review

[image]
[image]

peries, 2017-11-20:
@mnacinopa and I haven't been able to figure out how to properly disable it. Could one of you show us?

victorevector, 2017-11-22:
## Internal notes 

- how to reset password: https://github.com/foosel/OctoPrint/issues/1063

victorevector, 2017-11-22:
## SSH info

- An advanced user wants to enable SSH (default: disabled).

- A user wants to know the implications of turning it on.
    - Any remote user with the printer's username and password can control the printer. 

- A user wants to know the default password.
     - username: pi password: <printer-name>

- A user wants to know how to change the default password. 
    - https://www.raspberrypi.org/documentation/linux/usage/users.md

peries, 2017-11-28:
@victorevector @AllenMcAfee 
[image]
_________________________
[image]
_________________________
[image]
_________________________
[image]

peries, 2017-11-28:
and then if they've already created an account,
[image]

peries, 2017-11-29:
This is just an idea for new styling, although this content can be fitted into the current styling just fine. I can also send you the text in another document so you can copy/paste

peries, 2017-11-29:
@victorevector questions

1. can the 'access control' screens be dynamic such that it knows if you've created an admin account already?
2. can the 'setup complete' screen be dynamic such that it shows which selections you've just made (so you can confirm) before finishing the wizard?

victorevector, 2017-11-29:
@peries 

re: https://github.com/Robo3D/roboOctoprint/issues/13#issuecomment-347709584

1. The setup wizard does display a different interface if an admin account has already been registered.

2. I can carry over the values to the end screen. It does raise questions about "confirmation". Once you enable access control you can't undo it. Another thing, I think the surest way to reset an admin password is to reflash the whole OS.

mnacinopa, 2017-12-13:
@peries 
The only things to edit
- [ ] The default username is actually pi, I think?
- [ ] At the Enable/Disable Screens we need to warn them this will restart their printer and interrupt any print jobs going on

mnacinopa, 2017-12-13:
@victorevector 
When there are multiple user accounts can only the admin acc enable/disable access control & ssh?

peries, 2017-12-14:
@victorevector feedback for the latest version

Within the wizard, 
## Start
- [ ] second paragraph copy edit: If you're a new user, we'd like to welcome you to the Robo family! If you've been here before, we'll take you through some newly added settings.
- [ ] delete previous button
- [ ] delete 'unless otherwise noted..." line
- [ ] change next button text to 'start'

## Access Control
- [ ] Note copy edit: **NOTE:** If you forget your login credentials, you will need to contact Robo customer support to reset your password.
- [ ] delete previous button
- [ ] delete 'unless otherwise noted..." line

## SSH Control
- [ ] Change sidebar to say 'SSH Control' instead of wizard
- [ ] Change title to say 'SSH Control' 
- [ ] Delete apostrophe in it's
- [ ] Delete extra space after einstein
- [ ] Add period after closed parentheses
- [ ] Copy edit to the last line of the last paragraph: If you keep SSH disabled, there is no need to change your password.
- [ ] delete 'unless otherwise noted..." line
- [ ] change previous button text to 'go back'

## Finish
- [ ] delete 'unless otherwise noted..." line
- [ ] change previous button text to 'go back'

peries, 2017-12-14:
## Enabling
[image]

Title: Enable Security Features
Body: You are about to launch the setup wizard to enable remote access security features. You will have to complete the wizard in order to continue using the web dashboard. The dashboard and your printer will automatically restart once the wizard is finished.
Are you sure you want to proceed?

## Disabling
[image]

Title:
Body: You are about to disable the remote access security features and DELETE all user accounts. This will make your printer remotely accessible to everyone within your local network. The dashboard and your printer will automatically restart.

Are you sure you want to proceed?

peries, 2017-12-14:
Can we change the text of the errors that this screen reports?
[image]

Title: Enter valid username and password
Body: Please properly fill out the username and password forms
Invalid username
Invalid password
Passwords do not match

peries, 2017-12-14:
@victorevector  let's talk more about this notification when you can

[image]

victorevector, 2017-12-14:
@peries @mnacinopa @AllenMcAfee @Ximidar 

I am currently adding the SSH enable/disable radio button to `Settings > Access Control`. This made sense to me given that the option is originally bundled with Access Control in the Security Setup Wizard. The only caveat is that you can only see `Settings > Access Control` if you've enabled Access Control. What are your thoughts?

Pros:
+ Added security layer in front of SSH. You must be logged in with an admin account to enable/disable SSH

Cons:
- The default state (disabled) cannot be changed unless you run the Security Setup Wizard OR you enable Access Control

Ximidar, 2017-12-14:
If we had an easier way of enabling access control It wouldn't be a problem either way. I believe they should only see the SSH option when they have access control enabled. having that option hidden behind a login would be nice.

mnacinopa, 2017-12-14:
Will the Security Setup Wizard run automatically once someone navigates to the web dash after we push the update? The only way i can see this being a problem is for users who actually use SSH, and get blindsided.

victorevector, 2017-12-14:
@mnacinopa The user will have to enable the Security Setup Wizard from `System` menu drop down

mnacinopa, 2017-12-14:
@victorevector Okay. So once we push the update, will SSH just automatically be disabled for all users?

victorevector, 2017-12-14:
@mnacinopa Per Allen's request yes

victorevector, 2017-12-14:
@peries @mnacinopa @AllenMcAfee @Ximidar
I won't be able to add the SSH enable/disable button the Settings today. I've tried for a couple hours now and I don't have the proper environment to test it out.  I'll have something ready once I'm back in the office next week.

peries, 2017-12-22:
@victorevector 

1. Access control screen in the wizard, there's a couple typos
* Since we didn't explain the QR code and mobile app, can you change the first paragraph to
When Access Control is enabled, **you cannot control the printer from the web dashboard unless you are logged in.** Additionally, you can only access the API QR code from the logged in view of the web dashboard. This prevents unregistered users from accessing and using your printer via the internet or mobile app. We advise enabling Access Control when your printer is connected to an unsecure or shared network (such as a school or public places).
* Can you change the note to say
**NOTE:** If you lose your new password, you will need to contact Robo customer support. In which case, you risk losing all of the data on your printer.

2. When access control is enabled, the text for the button to disable it is gone
[image]

3. The password to SSH is raspberry. Is this gonna change?

4. for the notification
[image]
* we need to link that top line to the article that explains access control. Michael will make this once we make those changes to the lines in the access control wizard
* How to activate? Go to Settings, then 'Enable remote access security feature' to launch the setup wizard.
* What's inside this feature? Restrict control of the printer via internet access (web dashboard, mobile app, SSH)

AllenMcAfee, 2018-01-04:
@peries can we go ahead and make the link that is needed in the announcement window?  I'll need to get that into this fix.

peries, 2018-01-04:
@mnacinopa let us know when the article is updated

peries, 2018-01-05:
@AllenMcAfee https://help.robo3d.com/hc/en-us/articles/115002840791

peries, 2018-01-08:
Couple more things where the text is highlighted:
[image]
should say 'or the Robo mobile app'

[image]
should say 'password is 'raspberry''

Ximidar, 2018-01-08:
@peries I fixed those two edits. Could you pull and test?

peries, 2018-01-09:
Thanks Matt! Looks like everything has been resolved

closed 2018-01-19 by Ximidar