Robo3D/roboOctoprint #14: [Bug] Canceling during start gcode will cause a bed crash

Ximidar, 2017-12-02:
Canceling in the middle of start GCODE causes octoprint to inject the Cancel GCODE commands into the buffer. This causes the printer to be set up incorrectly and results in a bed crash.

```
Changing monitoring state from 'Operational' to 'Printing'
Recv: echo:busy: processing
Changing monitoring state from 'Printing' to 'Operational'
Recv: echo:busy: processing
Recv: X:0.00 Y:197.00 Z:25.12 E:0.00 Count X:0 Y:15768 Z:20104
Recv: ok
Send: G28 Z0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:0.00 Y:197.00 Z:260.00 E:0.00 Count X:0 Y:15768 Z:208062
Recv: ok
Send: G90
Recv: ok
Send: G92 E0
Recv: X:0.00 Y:197.00 Z:260.00 E:0.00 Count X:0 Y:15768 Z:208062
Recv: ok
Send: M84
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: M400
Recv: ok
Send: M220 S100
Recv: ok
Send: M220 S100
Recv: ok
Send: M221 S100
Recv: ok
Send: M106 S0
Recv: ok
Send: M221 S100
Recv: ok
Send: M106 S0
Recv: ok
Send: G21
Recv: ok
Send: G90
Recv: ok
Send: G92 E0
Recv: X:0.00 Y:197.00 Z:260.00 E:0.00 Count X:0 Y:15768 Z:208062
Recv: ok
Send: M114
Recv: X:0.00 Y:197.00 Z:260.00 E:0.00 Count X:0 Y:15768 Z:208062
Recv: ok
Send: M107
Recv: ok
Send: M104 S230
Recv: ok
Send: M140 S60
Recv: ok
Send: G36
Recv: X:0.00 Y:197.00 Z:260.00 E:0.00 Count X:0 Y:15768 Z:208062
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Probe Bounce is Z: 9.17
Recv: Probe Offset is Z: -4.02
Recv: Position After Adjustment X:8.00 Y:0.00 Z:10.00 E:0.00 Count X:640 Y:0 Z:8002
Recv: echo:Settings Stored (498 bytes; crc 9391)
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:184.01 Y:156.00 Z:9.02 E:0.00 Count X:14727 Y:12486 Z:11378
Recv: ok
Send: G4 S2
Recv: ok
Send: G1 Z15 F300
Recv: ok
Send: G1 X10 Y1 F7200
Recv: ok
Send: M109 S230
Recv: ok
Send: M140 S60
############################################################
# Right here the Start Gcode is interrupted and the Cancel GCODE begins
###########################################################
Recv: ok
Send: M220 S100
Recv: ok
Send: M221 S100
Recv: ok
Send: M106 S0
Recv: ok
Send: M104 S0
Recv: ok
Send: M140 S0
Recv: ok
Send: M107
Recv: ok
Send: G91
Recv: ok
Send: G1 Z10
Recv: ok
Send: G90
Recv: ok
Send: G28 X0 Y0
###########################################################
# Right here the cancel command Homes the X and Y which destroys our leveling data
###########################################################
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:0.00 Y:197.00 Z:25.01 E:0.00 Count X:0 Y:15768 Z:20014
Recv: ok
Send: G28 Z0
##########################################################
# Right here the command to home the Z axis is given, but the printer does not move the bed down for some reason.
##########################################################
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Changing monitoring state from 'Operational' to 'Printing'
Recv: echo:busy: processing
Recv: X:0.00 Y:197.00 Z:260.00 E:0.00 Count X:0 Y:15768 Z:208062
Recv: ok
Send: N1 G90*17
Recv: ok
Send: N2 G92 E0*69
Recv: X:0.00 Y:197.00 Z:260.00 E:0.00 Count X:0 Y:15768 Z:208062
Recv: ok
Send: N3 M84*28
Recv: ok
####################################################
# Right here the Start GCODE Resumes and runs into the bed
####################################################
Send: N4 G1 Z0.35*78
Recv: ok
Send: N5 G1 X190 E15.0 F500*65
Recv: ok
Send: N6 G1 Z15 F300*5
Recv: ok
Send: N7 G92 E0*64
Recv: X:190.00 Y:197.00 Z:15.00 E:0.00 Count X:0 Y:15768 Z:207449
Recv: ok
Send: N8 G1 F7200*67
Recv: ok
Send: N9 M105*46
```

Thanks @peries for getting the serial data.

peries, 2017-12-04:
```Recv:  T:230.19 /230.00 B:54.18 /60.00 @:52 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.00 /230.00 B:54.24 /60.00 @:56 B@:127 W:?
Recv:  T:229.94 /230.00 B:54.28 /60.00 @:58 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.00 /230.00 B:54.39 /60.00 @:56 B@:127 W:?
Recv:  T:230.06 /230.00 B:54.46 /60.00 @:54 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.13 /230.00 B:54.55 /60.00 @:53 B@:127 W:?
Recv:  T:230.06 /230.00 B:54.58 /60.00 @:54 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.94 /230.00 B:54.66 /60.00 @:57 B@:127 W:?
Recv:  T:230.00 /230.00 B:54.73 /60.00 @:55 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.06 /230.00 B:54.83 /60.00 @:53 B@:127 W:?
Recv:  T:230.00 /230.00 B:54.90 /60.00 @:55 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.00 /230.00 B:54.95 /60.00 @:55 B@:127 W:?
Recv:  T:229.94 /230.00 B:55.00 /60.00 @:56 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.00 /230.00 B:55.23 /60.00 @:55 B@:127 W:?
Recv:  T:230.06 /230.00 B:55.25 /60.00 @:53 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.94 /230.00 B:55.46 /60.00 @:57 B@:127 W:?
Recv:  T:229.88 /230.00 B:55.48 /60.00 @:58 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.94 /230.00 B:55.66 /60.00 @:57 B@:127 W:?
Recv:  T:230.00 /230.00 B:55.79 /60.00 @:55 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.94 /230.00 B:55.92 /60.00 @:57 B@:127 W:?
Recv:  T:229.94 /230.00 B:56.00 /60.00 @:57 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.88 /230.00 B:56.09 /60.00 @:59 B@:127 W:?
Recv:  T:230.13 /230.00 B:56.25 /60.00 @:53 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.13 /230.00 B:56.38 /60.00 @:53 B@:127 W:?
Recv:  T:230.00 /230.00 B:56.50 /60.00 @:56 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.00 /230.00 B:56.61 /60.00 @:56 B@:127 W:?
Recv:  T:229.94 /230.00 B:56.69 /60.00 @:57 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.00 /230.00 B:56.91 /60.00 @:55 B@:127 W:?
Recv:  T:229.94 /230.00 B:56.97 /60.00 @:57 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.94 /230.00 B:57.07 /60.00 @:57 B@:127 W:?
Recv:  T:230.00 /230.00 B:57.20 /60.00 @:55 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.00 /230.00 B:57.37 /60.00 @:55 B@:127 W:?
Recv:  T:229.88 /230.00 B:57.43 /60.00 @:58 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.81 /230.00 B:57.52 /60.00 @:60 B@:127 W:?
Recv:  T:229.88 /230.00 B:57.62 /60.00 @:58 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.63 /230.00 B:57.78 /60.00 @:64 B@:127 W:?
Recv:  T:229.88 /230.00 B:57.88 /60.00 @:57 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.94 /230.00 B:57.99 /60.00 @:56 B@:127 W:?
Recv:  T:229.94 /230.00 B:58.14 /60.00 @:56 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.81 /230.00 B:58.27 /60.00 @:59 B@:127 W:?
Recv:  T:229.94 /230.00 B:58.31 /60.00 @:56 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.06 /230.00 B:58.52 /60.00 @:53 B@:127 W:?
Recv:  T:229.94 /230.00 B:58.54 /60.00 @:57 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:230.06 /230.00 B:58.70 /60.00 @:53 B@:127 W:?
Recv:  T:229.88 /230.00 B:58.85 /60.00 @:58 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:229.94 /230.00 B:58.93 /60.00 @:57 B@:127 W:?
Recv:  T:230.13 /230.00 B:59.05 /60.00 @:53 B@:127 W:9
[... 497 lines cut ...]
Recv:  T:130.56 /0.00 B:53.37 /60.00 @:0 B@:127 W:?
Recv:  T:129.92 /0.00 B:53.40 /60.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:129.34 /0.00 B:53.60 /60.00 @:0 B@:127 W:?
Recv:  T:128.72 /0.00 B:53.64 /60.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:128.10 /0.00 B:53.72 /60.00 @:0 B@:127 W:?
Recv:  T:127.46 /0.00 B:53.69 /60.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:126.87 /0.00 B:53.89 /60.00 @:0 B@:127 W:?
Recv:  T:126.13 /0.00 B:53.83 /60.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:125.49 /0.00 B:53.90 /60.00 @:0 B@:127 W:?
Recv:  T:124.90 /0.00 B:53.97 /60.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:124.32 /0.00 B:53.81 /60.00 @:0 B@:127 W:?
Recv:  T:123.71 /0.00 B:54.02 /60.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:123.14 /0.00 B:54.01 /60.00 @:0 B@:127 W:?
Recv:  T:122.53 /0.00 B:54.36 /60.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:121.99 /0.00 B:54.23 /60.00 @:0 B@:127 W:?
Recv:  T:121.38 /0.00 B:54.29 /60.00 @:0 B@:127 W:?
 Advanced options
```

Ximidar, 2017-12-04:
Alright, this is still interrupting the previous script. I'll work on another solution @peries

Ximidar, 2017-12-04:
The problem is that Octoprint is not deleting the Queue when firing off the cancel print script. I added this function:
```
def kill_all_remaining_tasks(self):
		while not self._command_queue.empty():
			item = self._command_queue.get()
			self._logger.info("Deleting Task: " + str(item))
			self._command_queue.task_done()

		self._command_queue.join()
		return
```
That seems to kill of any remaining tasks before firing the cancel GCODE script.

Ximidar, 2017-12-04:
@peries I loaded the new octoprint on your machine. I'm not seeing the problem anymore, could you test and confirm that the problem is gone?

Ximidar, 2017-12-07:
Fixed.

closed 2017-12-07 by Ximidar