Robo3D/roboOctoprint #15: filament wizard not working after update v2.0

peries, 2017-12-07:
```
pi@laughing-lobster:~ $ tail -f .octoprint/logs/octoprint.log
2017-12-07 21:58:46,819 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 Z15 F300
2017-12-07 21:58:46,820 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 X10 Y1 F7200
2017-12-07 21:58:46,821 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: M109 S230
2017-12-07 21:58:46,823 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: M140 S60
2017-12-07 21:58:46,824 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 Z0.35
2017-12-07 21:58:46,825 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 X190 E15.0 F500
2017-12-07 21:58:46,826 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 Z15 F300
2017-12-07 21:58:46,827 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G92 E0
2017-12-07 21:58:46,828 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 F7200
2017-12-07 21:58:50,274 - octoprint.plugins.octoalexa - INFO - State is now: printing
2017-12-07 22:12:50,607 - kivy - INFO - Stopping monitor! ########################
2017-12-07 22:12:50,623 - octoprint.util.comm - INFO - Script: afterPrintPaused Reading Line: G1 Z33.53
2017-12-07 22:12:50,613 - kivy - INFO - Killed Child
2017-12-07 22:12:51,220 - octoprint.plugins.octoalexa - INFO - State is now: print paused
2017-12-07 22:12:55,551 - kivy - INFO - Stopping monitor! ########################
2017-12-07 22:12:55,554 - kivy - INFO - Killed Child
2017-12-07 22:12:55,594 - kivy - INFO - Process 1267 Starting for Pin Monitor
2017-12-07 22:12:55,598 - kivy - INFO - Pin Monitor Started #################
2017-12-07 22:12:55,617 - octoprint.util.comm - INFO - Script: beforePrintResumed Reading Line: G1 Z3.53
2017-12-07 22:12:56,163 - octoprint.plugins.octoalexa - INFO - State is now: printing
2017-12-07 22:13:10,601 - kivy - INFO - Total: 377551 1s: 377551 0s: 0
2017-12-07 22:13:10,603 - kivy - INFO - 1 Percentage: 100.0000
2017-12-07 22:13:10,605 - kivy - INFO - 0 Percentage: 0.0000
2017-12-07 22:13:10,606 - kivy - INFO - Pausing
2017-12-07 22:13:10,609 - kivy - INFO - pin monitor Stopped #################
2017-12-07 22:13:10,707 - kivy - INFO - True Pin Monitor Data ###########
2017-12-07 22:13:10,709 - kivy - INFO - Auto Pause Equals True!
2017-12-07 22:13:10,710 - kivy - INFO - Auto Paused
2017-12-07 22:13:10,911 - kivy - INFO - fil_runout
2017-12-07 22:13:11,358 - octoprint.util.comm - INFO - Script: afterPrintPaused Reading Line: G1 Z33.53
2017-12-07 22:13:11,356 - kivy - INFO - Stopping monitor! ########################
2017-12-07 22:13:11,626 - kivy - INFO - Killed Child
2017-12-07 22:13:14,257 - octoprint.plugins.octoalexa - INFO - State is now: print paused
2017-12-07 21:56:38,608 - octoprint.plugins.firmwareupdater - ERROR - Unable to get printer's data
2017-12-07 21:56:38,778 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.1.198/'}
2017-12-07 21:56:46,964 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.1.157/'}
2017-12-07 21:58:28,245 - kivy - INFO - Exiting Loading loop
2017-12-07 21:58:33,332 - kivy - INFO - Made temp directory for slicing
2017-12-07 21:58:33,454 - kivy - INFO - slicing wizard
2017-12-07 21:58:34,247 - kivy - INFO - setting support to: none
2017-12-07 21:58:34,393 - kivy - INFO - support_page
2017-12-07 21:58:46,692 - kivy - INFO - 3D_Benchy.gcode
2017-12-07 21:58:46,756 - kivy - INFO - Stopping monitor! ########################
2017-12-07 21:58:46,766 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G21
2017-12-07 21:58:46,768 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G90
2017-12-07 21:58:46,786 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G92 E0
2017-12-07 21:58:46,811 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: M107
2017-12-07 21:58:46,812 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: M104 S230
2017-12-07 21:58:46,814 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: M140 S60
2017-12-07 21:58:46,816 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G36
2017-12-07 21:58:46,817 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G4 S2
2017-12-07 21:58:46,819 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 Z15 F300
2017-12-07 21:58:46,820 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 X10 Y1 F7200
2017-12-07 21:58:46,821 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: M109 S230
2017-12-07 21:58:46,823 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: M140 S60
2017-12-07 21:58:46,824 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 Z0.35
2017-12-07 21:58:46,825 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 X190 E15.0 F500
2017-12-07 21:58:46,826 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 Z15 F300
2017-12-07 21:58:46,827 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G92 E0
2017-12-07 21:58:46,828 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 F7200
2017-12-07 21:58:50,274 - octoprint.plugins.octoalexa - INFO - State is now: printing
pi@laughing-lobster:~ $ tail -f .octoprint/logs/octoprint.log
2017-12-07 21:58:46,819 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 Z15 F300
2017-12-07 21:58:46,820 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 X10 Y1 F7200
2017-12-07 21:58:46,821 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: M109 S230
2017-12-07 21:58:46,823 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: M140 S60
2017-12-07 21:58:46,824 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 Z0.35
2017-12-07 21:58:46,825 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 X190 E15.0 F500
2017-12-07 21:58:46,826 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 Z15 F300
2017-12-07 21:58:46,827 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G92 E0
2017-12-07 21:58:46,828 - octoprint.util.comm - INFO - Script: beforePrintStarted Reading Line: G1 F7200
2017-12-07 21:58:50,274 - octoprint.plugins.octoalexa - INFO - State is now: printing
2017-12-07 22:12:50,607 - kivy - INFO - Stopping monitor! ########################
2017-12-07 22:12:50,623 - octoprint.util.comm - INFO - Script: afterPrintPaused Reading Line: G1 Z33.53
2017-12-07 22:12:50,613 - kivy - INFO2017-12-07 22:18:55,421 - kivy - INFO - Sto                                                                                                                                                             pping monitor! ########################
2017-12-07 22:18:55,449 - octoprint.util.comm - INFO - Script: afterPrintCancell                                                                                                                                                             ed Reading Line: M104 S0
2017-12-07 22:18:55,440 - kivy - INFO - Killed Child
2017-12-07 22:18:55,481 - octoprint.util.comm - INFO - Script: afterPrintCancell                                                                                                                                                             ed Reading Line: M140 S0
2017-12-07 22:18:55,506 - octoprint.util.comm - INFO - Script: afterPrintCancell                                                                                                                                                             ed Reading Line: M107
2017-12-07 22:18:55,515 - octoprint.util.comm - INFO - Script: afterPrintCancell                                                                                                                                                             ed Reading Line: G91
2017-12-07 22:18:55,522 - octoprint.util.comm - INFO - Script: afterPrintCancell                                                                                                                                                             ed Reading [... cut ...]
2017-12-07 22:18:56,114 - octoprint.plugins.octoalexa - INFO - State is now: rea                                                                                                                                                             dy
2017-12-07 22:18:56,569 - kivy - INFO - Cancellation: Successful
2017-12-07 22:18:56,707 - kivy - INFO - Stopping monitor! ######################                                                                                                                                                             ##
2017-12-07 22:18:56,726 - kivy - INFO - Killed Child

```

Ximidar, 2017-12-08:
It works on both our machines now for some reason

peries, 2017-12-12:
This is working now

Ximidar, 2017-12-12:
cool

closed 2017-12-12 by Ximidar