Robo3D/roboOctoprint #16: [Bug] re-enabling access control after disabling it results in a two step wizard

Ximidar, 2017-12-07:
[image]
[image]

When I disabled access control I deleted the `salt` entry under Access control, then added the `enabled` entry with a false value. then restarted octoprint. Everything went back to normal

When I re-enabled access controls I set the `enabled` tag to true, then set the `first start` tag to true. Then I restarted octoprint and the wizard looks like this.

Ximidar, 2017-12-07:
Then after I hit finish, it logs me in as Ximidar, but only because it remembers me. In incognito, it says user unkown.

### Normal Chrome
[image]
### Incognito mode
[image]

victorevector, 2017-12-11:
We already discussed this in person, but I'm documenting our discussion here...

This occurs because you disabled Access Control but didn't reset it. You must delete `/home/pi/.octprint/users.yaml` to remove any previous history and reset the wizard. Otherwise, octoprint still thinks there is a user.

closed 2017-12-11 by victorevector