Robo3D/roboOctoprint #19: API key does not always populate right away

peries, 2018-01-26:
#### What were you doing?

When Access Control is enabled, in order to add your printer to your mobile app, you need to scan the QR code under Settings > API.

#### What did you expect to happen?
To scan the QR code when it first opens

#### What happened instead?
The QR code displays but the API key was not populated. If the key reads n/a then you can't scan. You have to open and close the Settings a few times in order for the key to populate correctly.

#### Printer model & used firmware incl. version
R2, Robo 2.0, Marlin 1.2.3

#### Browser and Version of Browser, Operating System running Browser
Chrome, Windows 10

#### Screenshot(s) showing the problem:

see n/a in the API key textbox

[image]

Ximidar, 2018-01-30:
This may be my fault on your test image. I delete the API key then supposedly it should be generated on start up. I'll start generating a custom Key for your test images and see if that fixes it. Does it do this every time or only when your are testing it the first time?

peries, 2018-01-31:
it's intermittent and it's occurred on multiple images

Ximidar, 2018-01-31:
So it happens multiple times on the same image?

mnacinopa, 2018-01-31:
This happens on laughing-lobster every time you restart (Same image). The first time you try to open up the API key it doesn't populate. You have to press the refresh button >> Confirm >> reopen the settings to get it to populate

Ximidar, 2018-01-31:
is it the same API key or does it populate a different one every time?