Robo3D/roboOctoprint #22: [BUG] Access control will mess up plugins after it is enabled

Ximidar, 2018-02-03:
@peries enabled access control and then started a print with no filament. The filament sensor failed to start. Rebooting fixed the problems with the Filament Sensor. Access control may be messing with other plugins as well. Instead of just reloading, we should reboot the system after access control is enabled to avoid any unintentional errors.

peries, 2018-02-05:
@Ximidar the problem happened when I disabled access control and tried to run the first print. When it was enabled it was working fine

Ximidar, 2018-02-13:
@peries @mnacinopa  I updated the Filament sensor on your machines, If you could test it that would be great

Ximidar, 2018-02-13:
This bug has more to do with how we shut down the Octoprint instance. Some plugins have threads and processes that spawned, but are not controlled when it's time to shut down. The threads will eventually shut down, but the processes will hold on to the octoprint instance.